hello-robot/stretch_mujoco #25: Frozen runpy

Matthew-Mao-gAIns, 2024-09-10:
When I try to run python3 -m stretch_mujoco.stretch_mujoco.py, I get: 
<frozen runpy>:128: RuntimeWarning: 'stretch_mujoco.stretch_mujoco' found in sys.modules after import of package 'stretch_mujoco', but prior to execution of 'stretch_mujoco.stretch_mujoco'; this may result in unpredictable behaviour

snoato, 2024-09-10:
I get a similar error on macOS:
``` python3 -m stretch_mujoco.stretch_mujoco
<frozen runpy>:128: RuntimeWarning: 'stretch_mujoco.stretch_mujoco' found in sys.modules after import of package 'stretch_mujoco', but prior to execution of 'stretch_mujoco.stretch_mujoco'; this may result in unpredictable behaviour
2024-09-10 10:21:00.849 Python[36941:4079039] *** Assertion failure in -[NSMenu _setMenuName:], NSMenu.m:777
2024-09-10 10:21:00.850 Python[36941:4079039] *** Assertion failure in -[NSMenu _setMenuName:], NSMenu.m:777```

tjvitchutripop, 2024-10-09:
I'm having the same issue on macOS as well (however, I get the RuntimeWarning but I don't get the Assertion Failure; it just freezes after the RuntimeWarning).  After doing a bit of debugging, I think it has something to do with the threading in the start() function.  Not sure what's causing the issue though.

hello-binit, 2024-10-11:
Hey @tjvitchutripop and others, thanks for reporting this. @hello-atharva is going to see if he can reproduce this issue on his Mac. Can you share your OS version and chip architecture?

tjvitchutripop, 2024-10-11:
@hello-binit I'm using macOS Sequoia on an M3 Pro

hello-atharva, 2024-10-21:
Hi folks!

Could you try running my code that lives in this [branch](https://github.com/hello-robot/stretch_mujoco/pull/29)?

hello-binit, 2024-11-04:
Closing this since @hello-atharva's fix is merged. Feel free to open a new issue if it persists.

closed 2024-11-04 by hello-binit