hello-robot/stretch_mujoco #31: python3 -m stretch_mujoco

ramkumarkoppu, 2024-12-04:
when I run `python3 -m stretch_mujoco` don't see any output, the script is waiting forever seems
`robocasa` and `robosuite `are outside of `stretch_mujoco` directory, but all the dependent python packages are installed in the single conda environment called `stretch_mujoco_env` on Mac:
```
(stretch_mujoco_env) bash-5.2$ python --version
Python 3.10.15
```

hello-binit, 2024-12-04:
I see, thanks. Would you clone this repo and run 

```
# first remove stretch_mujoco from your conda env
cd stretch_mujoco/stretch_mujoco
# modify stretch_mujoco.py with the diff below
python stretch_mujoco.py
```

---

Modify with the following diff:

```diff
 if __name__ == "__main__":
-    import warnings
-    warnings.warn("use 'python -m stretch_mujoco', not 'python -m stretch_mujoco.stretch_mujoco'", DeprecationWarning)
-
-    # Check if we are on macOS
-    if os.uname().sysname == "Darwin":
-        print("macOS detected. Please use the following command to run the simulator:")
-        print("python3 -m stretch_mujoco")
-    else:
-        main()
+    main()
```

ramkumarkoppu, 2024-12-04:
still doesn't work:

```
(stretch_mujoco_env) Mac:stretch_mujoco ramkumarkoppu$ pip show stretch_mujoco
WARNING: Package(s) not found: stretch_mujoco
(stretch_mujoco_env) Mac:stretch_mujoco ramkumarkoppu$ git diff
diff --git a/stretch_mujoco/stretch_mujoco.py b/stretch_mujoco/stretch_mujoco.py
index cf13f11..92e2a36 100644
--- a/stretch_mujoco/stretch_mujoco.py
+++ b/stretch_mujoco/stretch_mujoco.py
@@ -14,8 +14,8 @@ import mujoco.viewer
 import numpy as np
 from mujoco import MjData, MjModel
 
-import stretch_mujoco.config as config
-import stretch_mujoco.utils as utils
+import config
+import utils
 
[...]
 
 if __name__ == "__main__":
-    import warnings
-    warnings.warn("use 'python -m stretch_mujoco', not 'python -m stretch_mujoco.stretch_mujoco'", DeprecationWarning)
-
-    # Check if we are on macOS
-    if os.uname().sysname == "Darwin":
-        print("macOS detected. Please use the following command to run the simulator:")
-        print("python3 -m stretch_mujoco")
-    else:
-        main()
+    main()
(stretch_mujoco_env) Mac:stretch_mujoco ramkumarkoppu$ pwd
/Users/ramkumarkoppu/repos/Robot_models/stretch_mujoco/stretch_mujoco
(stretch_mujoco_env) Mac:stretch_mujoco ramkumarkoppu$ python stretch_mujoco.py 
Traceback (most recent call last):
  File "/Users/ramkumarkoppu/miniconda3/envs/stretch_mujoco_env/lib/python3.10/site-packages/pkg_resources/__init__.py", line 419, in get_provider
    module = sys.modules[moduleOrReq]
KeyError: 'stretch_mujoco'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/ramkumarkoppu/repos/Robot_models/stretch_mujoco/stretch_mujoco/stretch_mujoco.py", line 18, in <module>
    import utils
  File "/Users/ramkumarkoppu/repos/Robot_models/stretch_mujoco/stretch_mujoco/utils.py", line 12, in <module>
    models_path = pkg_resources.resource_filename("stretch_mujoco", "models")
  File "/Users/ramkumarkoppu/miniconda3/envs/stretch_mujoco_env/lib/python3.10/site-packages/pkg_resources/__init__.py", line 1384, in resource_filename
    return get_provider(package_or_requirement).get_resource_filename(
  File "/Users/ramkumarkoppu/miniconda3/envs/stretch_mujoco_env/lib/python3.10/site-packages/pkg_resources/__init__.py", line 421, in get_provider
    __import__(moduleOrReq)
  File "/Users/ramkumarkoppu/repos/Robot_models/stretch_mujoco/stretch_mujoco/stretch_mujoco.py", line 508, in <module>
    "--scene-xml-path", default=utils.default_scene_xml_path, help="Path to the scene xml file"
AttributeError: partially initialized module 'utils' has no attribute 'default_scene_xml_path' (most likely due to a circular import)
(stretch_mujoco_env) Mac:stretch_mujoco ramkumarkoppu$ 
```

walker591, 2025-03-06:
(mujuco_env) walker@walker:~/project/stretch_mujoco$ python3 -m stretch_mujoco.stretch_mujoco
/home/walker/miniconda3/envs/mujuco_env/lib/python3.9/runpy.py:127: RuntimeWarning: 'stretch_mujoco.stretch_mujoco' found in sys.modules after import of package 'stretch_mujoco', but prior to execution of 'stretch_mujoco.stretch_mujoco'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
/home/walker/project/stretch_mujoco/stretch_mujoco/stretch_mujoco.py:537: DeprecationWarning: use 'python -m stretch_mujoco', not 'python -m stretch_mujoco.stretch_mujoco'
  warnings.warn("use 'python -m stretch_mujoco', not 'python -m stretch_mujoco.stretch_mujoco'", DeprecationWarning)
Traceback (most recent call last):
  File "/home/walker/miniconda3/envs/mujuco_env/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/walker/miniconda3/envs/mujuco_env/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/walker/project/stretch_mujoco/stretch_mujoco/stretch_mujoco.py", line 544, in <module>
    main()
  File "/home/walker/miniconda3/envs/mujuco_env/lib/python3.9/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
  File "/home/walker/miniconda3/envs/mujuco_env/lib/python3.9/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
  File "/home/walker/miniconda3/envs/mujuco_env/lib/python3.9/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/walker/miniconda3/envs/mujuco_env/lib/python3.9/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
  File "/home/walker/project/stretch_mujoco/stretch_mujoco/stretch_mujoco.py", line 515, in main
    robot_sim = StretchMujocoSimulator(scene_xml_path)
  File "/home/walker/project/stretch_mujoco/stretch_mujoco/stretch_mujoco.py", line 39, in __init__
    self.mjmodel = mujoco.MjModel.from_xml_path(scene_xml_path)
ValueError: Error: for mesh geoms, inertia should be specified in the mesh asset
Element name '', id 2, line 0

hello，what is this problem?how can I fix it

hello-robot-shehab, 2025-03-13:
Hi @walker591, we noticed a few issues with running the simulation on a MacOS. We're working on fixing this now. I will update you when we have it ready for you to test. Thank you for your patience.

hello-robot-shehab, 2025-03-31:
This should be resolved by #32 and #38. Thanks.

hello-robot-shehab, 2025-04-01:
Resolved in latest `main` branch. Thanks!

closed 2025-04-01 by hello-robot-shehab