hello-robot/stretch_mujoco #35: Invalid robosuite branch robocasa_v0.1

AlfredMoore, 2025-02-18:
Hi,

Reference: https://github.com/hello-robot/stretch_mujoco?tab=readme-ov-file#install-robocasa

I am trying to deploy mujoco simulation and find some errors. The robocasa:main does not support robosuite:robocasa_v0.1 any more.

I tried the latest robosuite and find the following light edit work. 

To use the latest robocasa and robosuite, my suggestion is to update [robocase_gen.py line 10](https://github.com/hello-robot/stretch_mujoco/blob/main/stretch_mujoco/robocasa_gen.py#L10) with updated function
```python
from robosuite import load_part_controller_config
```
and update [README.md line 40](https://github.com/hello-robot/stretch_mujoco/blob/main/README.md?plain=1#L40) with master branch
```bash
git clone https://github.com/ARISE-Initiative/robosuite
```

hello-binit, 2025-02-21:
Thanks @AlfredMoore, I will try out your suggestions and make the changes soon.

hello-robot-shehab, 2025-03-31:
This should be resolved by #38. We moved to a new dependency manager ([uv](https://docs.astral.sh/uv/guides/install-python/)) and added instructions in the README to use Robocasa on both linux and macos.

hello-robot-shehab, 2025-04-01:
Resolved in latest `main` branch. Thanks!

closed 2025-04-01 by hello-robot-shehab