hello-robot/stretch_mujoco #36: GPU option for lower latency

AlfredMoore, 2025-03-02:
Hi,

I just find the latency for camera pulling and rendering is kind of high for cv-based simulation.

Each rgbd camera takes about 0.04s to render, and the total can be 0.12s - 0.15s.

Is there any option to save time in rendering? I tried to make it multi thread but it seems that openGL is not thread-safe.

I integrated the repo into ROS Humble. In script, the camera rendering could only exist in the main thread, which limits the ros spin rate (<10Hz). The streaming image in rviz also looks stiff due to latency.

Looking forward to hearing from you.

hello-robot-shehab, 2025-03-13:
Hi @AlfredMoore, thank you for your message. 

Short-term, if you're only using the RGB cameras, you can comment-out the `.render()` calls for the other cameras, which should significantly increase the performance.

fyi, we're working on improving the rendering time - I'll update you when we have something you could test. 

Can you tell us more about your running OS and computer specs, if you're using any GL options, any specific info about the ROS integration that could help us pinpoint how rendering is affecting your main thread (currently a separate Process is spun for simulation and rendering), and how you're measuring the 0.04 render time?
Also, in the title, you mention a GPU option, could you clarify what you mean by this? Are you talking about MJX or something else? 

Thanks!

AlfredMoore, 2025-03-18:
Thanks for you support!

Runtime specs:
 * Ubuntu 22.04, Docker Container for ROS2 Humble
 * CPU Intel i5 13600kf
 * GPU RTX4070S
 * openGL option: I tried both default option and `egl` but find no significant improvement in efficiency

ROS integration:
I followed the `stretch_ros2` to integrate the `stretch-mujoco` as a pseudo hardware interface. Just as `stretch_ros2`, the node is managed by an instance of executor.
At first, the render is instantialized in the main thread and I tried to pull render data from a ROS timer callback, but that callback creates another thread and causes thread collision.
Then I tried to modified the callback group into the main group but still failed somehow.
Finally, I extracted the simulator instance outside, and just passed it to the ROS node. In that case, the simulator and the render are in the main thread of Python, and then it eventually works.

For reference, here is [my repo](https://github.com/AlfredMoore/stretch-vr) for that but it is kind of messy inside. The main running script is [here](https://github.com/AlfredMoore/stretch_ros2/blob/8057dab97a065166d49a74aac69379b917d23bcd/stretch_mujoco_ros2/stretch_mujoco_ros2/stretch_sim_driver.py#L584).

Rendering time:
After commenting the render in simulator and do this
```python
t1 = time.time()
camera_data = robot_sim.pull_camera_data()
t2 = time.time()
node.get_logger().info(f"Time taken for camera rendering: {t2 - t1}")
```

GPU option:
I mean whether there is a way to improve rendering latency, both for openGL and MJX. Cuz I tried `egl` option for GL but found the GPU is not used. According to `nvidia-smi`
```bash
root@humble-container:/workspace# nvidia-smi 
Tue Mar 18 17:25:17 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.120                Driver Version: 550.120        CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4070 ...    Off |   00000000:01:00.0  On |                  N/A |
|  0%   46C    P5             18W /  220W |     712MiB /  12282MiB |     37%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
```

hello-robot-shehab, 2025-03-24:
Putting this here for GPU support. 

The Mujoco teams' Colab notebooks (https://colab.research.google.com/github/google-deepmind/mujoco_playground/blob/main/learning/notebooks/manipulation.ipynb#scrollTo=IbZxYDxzoz5R) have this cell to enable GPU rendering:

```
# @title Check if MuJoCo installation was successful

import distutils.util
import os
import subprocess

if subprocess.run('nvidia-smi').returncode:
  raise RuntimeError(
      'Cannot communicate with GPU. '
      'Make sure you are using a GPU Colab runtime. '
      'Go to the Runtime menu and select Choose runtime type.'
  )

# Add an ICD config so that glvnd can pick up the Nvidia EGL driver.
# This is usually installed as part of an Nvidia driver package, but the Colab
# kernel doesn't install its driver via APT, and as a result the ICD is missing.
# (https://github.com/NVIDIA/libglvnd/blob/master/src/EGL/icd_enumeration.md)
NVIDIA_ICD_CONFIG_PATH = '/usr/share/glvnd/egl_vendor.d/10_nvidia.json'
if not os.path.exists(NVIDIA_ICD_CONFIG_PATH):
  with open(NVIDIA_ICD_CONFIG_PATH, 'w') as f:
    f.write("""{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "libEGL_nvidia.so.0"
    }
}
""")

# Configure MuJoCo to use the EGL rendering backend (requires GPU)
print('Setting environment variable to use GPU rendering:')
%env MUJOCO_GL=egl

try:
  print('Checking that the installation succeeded:')
  import mujoco

  mujoco.MjModel.from_xml_string('<mujoco/>')
except Exception as e:
  raise e from RuntimeError(
      'Something went wrong during installation. Check the shell output above '
      'for more information.\n'
      'If using a hosted Colab runtime, make sure you enable GPU acceleration '
      'by going to the Runtime menu and selecting "Choose runtime type".'
  )

print('Installation successful.')

# Tell XLA to use Triton GEMM, this improves steps/sec by ~30% on some GPUs
xla_flags = os.environ.get('XLA_FLAGS', '')
xla_flags += ' --xla_gpu_triton_gemm_any=True'
os.environ['XLA_FLAGS'] = xla_flags
```

hello-robot-shehab, 2025-03-24:
Also, @AlfredMoore, a quick update: we have improved the soft-rendering performance by separating the physics stepping from the rendering using threading. Updates coming soon. Thanks.

hello-robot-shehab, 2025-03-31:
This should be resolved by #38. Thanks.

hello-robot-shehab, 2025-04-01:
Hey @AlfredMoore, could you please re-test with the latest StretchMujocoSimulator in `main`? You may need to add the `cameras_to_use` argument to your [`stretch_mujoco_ros2/stretch_mujoco_ros2/stretch_sim_driver.py`](https://github.com/AlfredMoore/stretch_ros2/blob/8057dab97a065166d49a74aac69379b917d23bcd/stretch_mujoco_ros2/stretch_mujoco_ros2/stretch_sim_driver.py#L584)
```
from stretch_mujoco.enums.stretch_cameras import StretchCameras
robot_sim = StretchMujocoSimulator(default_scene_xml_path, cameras_to_use=[StretchCameras.cam_d405_rgb])
```
 Thanks!

hello-robot-shehab, 2025-05-15:
Hey @AlfredMoore, 

Sorry, I totally forgot to measure and post these.

fyi, I did this profiling just now on the latest main branch.
I used a delta of sim.pull_camera_data().time -> this is sim time. I also did a diff using wall time. 
In short, I did not see much of a difference in camera rendering time between 1 and 5 cameras. However, the sim-to-real performance was affected instead with more compute intensive options. Using `export MUJOCO_GL=egl` makes a huge difference.

12th Gen Intel® Core™ i7-12700H 16GB RAM
GeForce RTX 3060 Mobile 6GB GPU memory

I used `export MUJOCO_GL=egl` before running these.

Here's 1 camera with camera_hz set to 30: 
```
Time taken for camera rendering: 0.019999999999997797. Sim is running 0.661x as fast as realtime
Time taken for camera rendering: 0.023999999999997357. Sim is running 0.661x as fast as realtime
Time taken for camera rendering: 0.021999999999997577. Sim is running 0.661x as fast as realtime
Time taken for camera rendering: 0.025999999999997137. Sim is running 0.661x as fast as realtime
Time taken for camera rendering: 0.021999999999997577. Sim is running 0.661x as fast as realtime
Time taken for camera rendering: 0.023999999999997357. Sim is running 0.661x as fast as realtime
Time taken for camera rendering: 0.023999999999997357. Sim is running 0.661x as fast as realtime
Time taken for camera rendering: 0.023999999999997357. Sim is running 0.692x as fast as realtime
Time taken for camera rendering: 0.021999999999997577. Sim is running 0.692x as fast as realtime
Time taken for camera rendering: 0.023999999999997357. Sim is running 0.692x as fast as realtime
Time taken for camera rendering: 0.021999999999997577. Sim is running 0.692x as fast as realtime
Time taken for camera rendering: 0.021999999999997577. Sim is running 0.692x as fast as realtime
```
The same with wall time:
```
Time taken for camera rendering: 0.03350136798690073. Sim is running 0.681x as fast as realtime
Time taken for camera rendering: 0.03336717700585723. Sim is running 0.681x as fast as realtime
Time taken for camera rendering: 0.03340022198972292. Sim is running 0.681x as fast as realtime
Time taken for camera rendering: 0.0334339480032213. Sim is running 0.681x as fast as realtime
Time taken for camera rendering: 0.03339132000110112. Sim is running 0.681x as fast as realtime
Time taken for camera rendering: 0.033440908999182284. Sim is running 0.681x as fast as realtime
Time taken for camera rendering: 0.03339212201535702. Sim is running 0.681x as fast as realtime
```

Here's 1 camera with camera_hz set to 100: 
```
Time taken for camera rendering: 0.005999999999999339. Sim is running 0.581x as fast as realtime
Time taken for camera rendering: 0.005999999999999339. Sim is running 0.581x as fast as realtime
Time taken for camera rendering: 0.005999999999999339. Sim is running 0.581x as fast as realtime
Time taken for camera rendering: 0.013999999999998458. Sim is running 0.581x as fast as realtime
Time taken for camera rendering: 0.027999999999996916. Sim is running 0.581x as fast as realtime
Time taken for camera rendering: 0.031999999999996476. Sim is running 0.581x as fast as realtime
Time taken for camera rendering: 0.005999999999999339. Sim is running 0.581x as fast as realtime
Time taken for camera rendering: 0.013999999999998458. Sim is running 0.581x as fast as realtime
Time taken for camera rendering: 0.005999999999999339. Sim is running 0.581x as fast as realtime
Time taken for camera rendering: 0.005999999999999339. Sim is running 0.581x as fast as realtime
Time taken for camera rendering: 0.005999999999999339. Sim is running 0.581x as fast as realtime
Time taken for camera rendering: 0.007999999999999119. Sim is running 0.581x as fast as realtime
```

Here's 5 camera with camera_hz set to 30: 
```
Time taken for camera rendering: 0.027999999999996916. Sim is running 0.363x as fast as realtime
Time taken for camera rendering: 0.025999999999997137. Sim is running 0.363x as fast as realtime
Time taken for camera rendering: 0.009999999999998899. Sim is running 0.363x as fast as realtime
Time taken for camera rendering: 0.011999999999998678. Sim is running 0.363x as fast as realtime
Time taken for camera rendering: 0.007999999999999119. Sim is running 0.363x as fast as realtime
Time taken for camera rendering: 0.017999999999998018. Sim is running 0.363x as fast as realtime
Time taken for camera rendering: 0.007999999999999119. Sim is running 0.363x as fast as realtime
```

the same with wall time:
```
Time taken for camera rendering: 0.04600881499936804. Sim is running 0.372x as fast as realtime
Time taken for camera rendering: 0.03343398199649528. Sim is running 0.372x as fast as realtime
[... 50 lines cut ...]
Here's 1 camera with camera_hz set to 30:
```
Time taken for camera rendering: 0.007999999999999119. Sim is running 0.332x as fast as realtime
Time taken for camera rendering: 0.011999999999998678. Sim is running 0.343x as fast as realtime
Time taken for camera rendering: 0.011999999999998678. Sim is running 0.343x as fast as realtime
Time taken for camera rendering: 0.007999999999999119. Sim is running 0.343x as fast as realtime
Time taken for camera rendering: 0.009999999999998899. Sim is running 0.343x as fast as realtime
Time taken for camera rendering: 0.009999999999998899. Sim is running 0.343x as fast as realtime
Time taken for camera rendering: 0.013999999999998458. Sim is running 0.343x as fast as realtime
[... cut ...]
Time taken for camera rendering: 0.0020000000000000018. Sim is running 0.289x as fast as realtime
Time taken for camera rendering: 0.0020000000000000018. Sim is running 0.289x as fast as realtime
Time taken for camera rendering: 0.019999999999999796. Sim is running 0.289x as fast as realtime
Time taken for camera rendering: 0.0019999999999997797. Sim is running 0.289x as fast as realtime
Time taken for camera rendering: 0.0039999999999995595. Sim is running 0.289x as fast as realtime
Time taken for camera rendering: 0.0019999999999997797. Sim is running 0.289x as fast as realtime
```

closed 2025-05-15 by hello-robot-shehab

AlfredMoore, 2025-05-16:
Thank you so much for supporting me. And it is so nice to know the real-time performance on camera rendering 🙂

Very helpful to my realtime simulation on Mujoco.

AlfredMoore, 2025-05-16:
> Of course! A few more updates are coming soon to improve overall usability and closeness to the real robot. Please reach out and create issues if you see anything that is helpful to you. Thank you!

[like]  Mo Xu reacted to your message: