hello-robot/stretch_mujoco #43: Disable lidar lines using `mjVIS_RANGEFINDER`

hello-robot-shehab, 2025-04-01:
Right now, we're hiding the lidar lines by setting the alpha value of the lines to 0; `self.mjmodel.vis.rgba.rangefinder[3] = 0`

The Mujoco team recommended using 
```
opt = mujoco.MjvOption()
opt.flags[mujoco.mjtVisFlag.mjVIS_RANGEFINDER] = False
```
to disable the visualization instead.  
https://github.com/google-deepmind/mujoco/discussions/2547#discussioncomment-12687275
https://mujoco.readthedocs.io/en/latest/APIreference/APItypes.html#mjtvisflag

I'm not sure if there's a performance difference, but it may save the viewer slight rendering time.

hello-binit, 2025-04-01:
Looks like the lidar lines are showing up in the depth image. This might be why they recommended that approach.

[image]

hello-robot-shehab, 2025-04-02:
Good catch! Trying to figure out the best way to plug-in those render options.

PR #83 (2025-06-06): Rangefinder lines are now disabled in Mujoco scene options
Resolves #43
closed 2025-06-06 by hello-robot-shehab