hello-robot/stretch_mujoco #46: `move_by` should give feedback on whether the requested position is reached (or thought to be reached).

hello-robot-shehab, 2025-04-02:
In `/stretch_mujoco_simulator.py`, the `move_by` command is a non-blocking function. When the user sends a `move_by` command, there is no blocking call or feedback whether the robot has moved. 

We need either a timeout (similar to `move_to`'s) to block until the robot reaches the position, or a different mechanism for the user to check if the desired position has been reached (without having to implement the logic themselves).

We also need to amend the docstrings since the `timeout` argument is not operational.

[image]

hello-binit, 2025-04-03:
The blocking `move_to()` is actually a departure from the real robot's API. For Stretch Body, `move_to()` is non-blocking, and there's a `wait_command()` method if you want to wait til completion. As you've noted, `move_by()` is non-blocking, and therefore matches the real API better.

hello-robot-shehab, 2025-05-15:
This should be resolved by #61.

PR #61 (2025-05-28): Adds the `wait_until_at_setpoint()` and `wait_while_is_moving()` methods available in StretchBody to wait for a joint to finish moving.
Resolves #46
closed 2025-05-28 by hello-robot-shehab