hello-robot/stretch_mujoco #58: TkAgg plotting error

hello-binit, 2025-04-26:
Running the laser scan example tries to plot with the Tkinter backend, but yields the following error:

```
$ uv run examples/laser_scan.py 
AttributeError: module '_tkinter' has no attribute '__file__'. Did you mean: '__name__'?

The above exception was the direct cause of the following exception:

ImportError: failed to load tkinter functions

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/binitshah/repos/stretch_mujoco/examples/laser_scan.py", line 5, in <module>
    matplotlib.use("TkAgg")
  File "/home/binitshah/repos/stretch_mujoco/.venv/lib/python3.10/site-packages/matplotlib/__init__.py", line 1281, in use
    plt.switch_backend(name)
  File "/home/binitshah/repos/stretch_mujoco/.venv/lib/python3.10/site-packages/matplotlib/pyplot.py", line 425, in switch_backend
    module = backend_registry.load_backend_module(newbackend)
  File "/home/binitshah/repos/stretch_mujoco/.venv/lib/python3.10/site-packages/matplotlib/backends/registry.py", line 317, in load_backend_module
    return importlib.import_module(module_name)
  File "/home/binitshah/.local/share/uv/python/cpython-3.10.17-linux-x86_64-gnu/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/binitshah/repos/stretch_mujoco/.venv/lib/python3.10/site-packages/matplotlib/backends/backend_tkagg.py", line 1, in <module>
    from . import _backend_tk
  File "/home/binitshah/repos/stretch_mujoco/.venv/lib/python3.10/site-packages/matplotlib/backends/_backend_tk.py", line 25, in <module>
    from . import _tkagg
ImportError: initialization failed
```

hello-binit, 2025-04-26:
I've tried using QT instead of Tkinter, with `matplotlib.use("Qt5Agg")` at

https://github.com/hello-robot/stretch_mujoco/blob/3d0f6d7e3eb2be34bf7a9f7e7f42af66af8b8316/examples/laser_scan.py#L5

and adding PyQt5 (`uv add PyQt5`), but this yields a different error:
```
QObject::moveToThread: Current thread (0x6250e8257a20) is not the object's thread (0x6250e896fa20).
Cannot move to target thread (0x6250e8257a20)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/binitshah/repos/stretch_mujoco/.venv/lib/python3.10/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl.
```
The sim window appears, but matplotlib doesn't.

hello-robot-shehab, 2025-04-26:
Could you please try commenting out ` matplotlib.use("TkAgg") `? Commenting it out makes it work on my mac. Thanks!