hello-robot/stretch_mujoco #95: Cannot download textures.zip

orange1238321, 2025-06-12:
uv run third_party/robocasa/robocasa/scripts/download_kitchen_assets.py

When I run this command, it fails to download textures.zip. Could you help me resolve this issue?

Downloading to /home/mycomputer/stretch_mujoco/third_party/robocasa/robocasa/models/assets/generative_textures.zip
generative_textures.zip: 0.00B [00:03, ?B/s]
Error downloading after try #3
Failed to download. Try again...

hello-robot-shehab, 2025-06-12:
Hi @orange1238321 , thank you for reaching out.  The textures and assets are pretty big (~5gb).

All the assets are hosted at `https://utexas.box.com/` by the robocasa team. Here's an excerpt from  `download_kitchen_assets.py`:
```
DOWNLOAD_ASSET_REGISTRY = dict(
    textures=dict(
        message="Downloading environment textures",
        url="https://utexas.box.com/shared/static/otdsyfjontk17jdp24bkhy2hgalofbh4.zip",
        folder=os.path.join(robocasa.__path__[0], "models/assets/textures"),
        check_folder_exists=False,
    ),
    fixtures=dict(
        message="Downloading fixtures",
        url="https://utexas.box.com/shared/static/pobhbsjyacahg2mx8x4rm5fkz3wlmyzp.zip",
        folder=os.path.join(robocasa.__path__[0], "models/assets/fixtures"),
        check_folder_exists=False,
    ),
    objaverse=dict(
        message="Downloading objaverse objects",
        url="https://utexas.box.com/shared/static/ejt1kc2v5vhae1rl4k5697i4xvpbjcox.zip",
        folder=os.path.join(robocasa.__path__[0], "models/assets/objects/objaverse"),
        check_folder_exists=False,
    ),
    aigen_objs=dict(
        message="Downloading AI-generated objects",
        url="https://utexas.box.com/shared/static/os3hrui06lasnuvwqpmwn0wcrduh6jg3.zip",
        folder=os.path.join(robocasa.__path__[0], "models/assets/objects/aigen_objs"),
        check_folder_exists=False,
    ),
    generative_textures=dict(
        message="Downloading AI-generated environment textures",
        url="https://utexas.box.com/shared/static/gf9nkadvfrowkb9lmkcx58jwt4d6c1g3.zip",
        folder=os.path.join(robocasa.__path__[0], "models/assets/generative_textures"),
        check_folder_exists=False,
    ),
)
```

Can you make sure your network can reach `box.com`? Thanks!

orange1238321, 2025-06-12:
Thank you! It seems like there is net connection wrong with Box.com, I will try it later!

hello-robot-shehab, 2025-06-12:
> Thank you! It seems like there is net connection wrong with Box.com, I will try it later!

I'm glad we found the cause of your problem. Note that access to box.com may be blocked by your network administrator. It may be helpful to reach out to your IT support for help.

closed 2025-06-12 by hello-binit